ENACRobotique/FAT_2018 #7: Make a reset function

Fabien-B, 2017-12-22:
To reset the teensy is gonna be a problem : the PCB seem to be not quite good as expected.
We should make a function to "reset" all relevant variables to their original state.
This function will be called either from the communication class, either as the result of a given logic level on a dedicated pin. Both is better.
In the case of the communication class, the message must be accepted no matter his id is.
This function could also set the next expected message id to n+1 where n is the message id of the reset message.

Fabien-B, 2017-12-22:
I handle the classes ExtNagivation, MotorControl, Odometry and Simulator.

guilhembn, 2017-12-22:
So there will be one more message type "RESET" that can be send from the raspberry pi. I implement the same callbacks system on this message too, so when a RESET message is received, it is accepted whatever its id, and the registered callbacks are called. I think it is better to reset the expected next message to 0 too.

guilhembn, 2017-12-22:
225371355854860b4d0e0fad712f8206a43557ac & 617d15ffa5c6e19fda99d8fb362e8dd5647f10ef

closed 2017-12-22 by guilhembn