ENACRobotique/FAT_2018 #13: Adds behavior selection on ai launch

guilhembn, 2018-01-10:
Let the user choose the wanted behaviour when launching main.py.

1. Put the behaviour register (the dictionary) in behaviour.__init__.py
2. Use arg parse to select one of the entry and instantiate the Robot object with this argument.
3. Enhance -h argument by showing all the behaviour available (by browsing the behaviour register).

Ex :
`$ python3 main.py -h
[...]
-b, --behaviour BEHAVIOR
    Available behviours :
        * FSMMatch
        * Slave
        * IvyJoystickSlave
        * PluggedXboxControllerSlave
[...]`
`$ python3 main.py -b Slave`

Fabien-B, 2018-01-10:
This argument should not be mandatory. If not present, a default behavior (hard coded) must be used.

guilhembn, 2018-01-10:
Of course, how can it be otherwise ? :stuck_out_tongue: